usfsoar/NSL_23-24_PayloadEECS #69: Connect 433MHz Receiver and Test it

L42ARO, 2024-01-22:
- [ ] Connect the 433MHz module to the ESP32S3, and get it to transmit.
- [ ] Document on the wiki how you got it to work

Pavan-da-Coder, 2024-01-26:
To connect with an Arduino Uno to an Arduino Uno,
Follow the wiring diagram EXACTLY as is shown
https://www.airspayce.com/mikem/arduino/RadioHead/classRH__RF95.html
[image]

Pavan-da-Coder, 2024-01-26:
Next task is to solder pins and connect the transceiver to the ESP32-S3

closed 2024-02-04 by L42ARO

Pavan-da-Coder, 2024-02-04:
Sample Code, condensed, from community
https://forums.adafruit.com/viewtopic.php?t=104190

Sample code, not condensed, from developers
https://www.airspayce.com/mikem/arduino/RadioHead/rf95_reliable_datagram_client_8pde-example.html
https://www.airspayce.com/mikem/arduino/RadioHead/rf95_reliable_datagram_server_8pde-example.html

Documentation:
https://www.airspayce.com/mikem/arduino/RadioHead/classRH__RF95.html#a274c170acdb53f55fd45445ab7f51889

Pavan-da-Coder, 2024-02-04:
Documentation parent links:
https://www.airspayce.com/mikem/arduino/RadioHead/
https://www.airspayce.com/mikem/arduino/RadioHead/examples.html